UncleRus/esp-idf-lib #706: Driver for SHTC3

xyzroe, 2025-06-09:
### Device name

Sensirion SHTC3

### Device description

±2% (20-80%RH) Digital humidity and temperature sensor for battery-driven applications

P.S. I have already implemented it. I need some time for testing, and then I will make a pull request.

### Device product page URL

https://sensirion.com/products/catalog/SHTC3

### Datasheet URL

https://sensirion.com/media/documents/643F9C8E/63A5A436/Datasheet_SHTC3.pdf

### Can the device be purchased at any of the followings?

- [x] AliExpress
- [ ] Amazon
- [ ] ebay
- [ ] Digi-Key
- [ ] Mouser

### URLs to purchase the device

https://www.aliexpress.com/item/1005005540209182.html

### Other implementations

https://github.com/mauriciobarroso/shtc3/

xyzroe, 2025-06-22:
Hello. 
Sorry, I haven't committed it to the repository yet. However, if you are interested, you can use this zip file.

[shtc3.zip](https://github.com/user-attachments/files/20852547/shtc3.zip)

Keep in mind that I haven't tested it yet. I'm still waiting for the sensor.

Basic example 
``` c
#include "shtc3.h"

#define SHTC3_I2C_ADDR 0x70
#define SHTC3_I2C_PORT I2C_NUM_0
#define SHTC3_SDA_GPIO 21
#define SHTC3_SCL_GPIO 22

void shtc3_task(void *pvParameter)
{
  shtc3_t shtc3;
  float temp = 0, hum = 0;

  if (shtc3_init(&shtc3, SHTC3_I2C_ADDR, SHTC3_I2C_PORT, SHTC3_SDA_GPIO, SHTC3_SCL_GPIO) != ESP_OK)
  {
    vTaskDelete(NULL);
  }

  while (1)
  {
    if (shtc3_get_temp_and_hum(&shtc3, &temp, &hum) == ESP_OK)
    {
      ESP_LOGI("shtc3_task", "Temp: %.2f C, Hum: %.2f %%", temp, hum);
    }
    else
    {
      ESP_LOGW("shtc3_task", "Failed to read SHTC3");
    }
    vTaskDelay(pdMS_TO_TICKS(2000));
  }
}

void app_main(void)
{
  const char *TAG = "main";
  i2cdev_init();
  xTaskCreate(shtc3_task, "shtc3_task", 2048, NULL, 1, NULL);
}

```

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/core/issues/23

closed 2025-08-01 by trombik