UncleRus/esp-idf-lib #719: [heads up] We are moving! No new issues here

trombik, 2025-07-24:
### The issue

We do not accept any issues anymore. Sorry. Instead, please wait until we finish the migration. We will announce once it completes.

You might have found a bug in our code and came here. However, we will move to an GitHub organization, [esp-idf-lib](https://github.com/esp-idf-lib). All the components in this repository will migrate, too. Until then, we stop accepting issues. When all the repositories are ready, please open an Issue in the relevant repository.

You MAY open an Issue here if the issue has nothing to do with the code.

### Benefits

* Each component has its own repository. Faster build, simpler CI, easier to add more components
* [ESP Component Registry](https://components.espressif.com/) integration. No need to import the whole repository.

This repository is owned by @UncleRus. Not mine. I have not been active on the project, but it seems like, I am the only one who can move the project forward. If you are willing to maintain the project, please help me. There are many remaining issues you can help.

### Confirmation

- [x] This report is not a bug report in code, a question, nor a request for drivers.

trombik, 2025-07-29:
@xyzroe 

You can open issues or PRs in the new repositories. I'll request updates in the remaining issues and PRs soon.

trombik, 2025-07-29:
@xyzroe 

Note that you need:

```console
git submodule update --init --recursive
```

after `git clone`.

xyzroe, 2025-07-29:
> 1. fork the component repository (NOT "core" repository)

What repository should I fork when I need to create a **new** repo for a **new** component?

trombik, 2025-07-29:
here is the work-in-progress: https://github.com/esp-idf-lib/template-component/blob/main/TEMPLATE_README.md

xyzroe, 2025-07-29:
Thanks. I'll try later.

trombik, 2025-07-30:
@xyzroe 

Here is one ported from a PR.
https://github.com/esp-idf-lib/bmi160/commits/main/

Hope it would not be complicated.