UncleRus/esp-idf-lib #721: pcf8574.c references non-existent clk_flags field in i2c_dev_t structure

Ligament, 2025-07-25:
### The issue

The `pcf8574.c` file attempts to set `dev->cfg.clk_flags = 0;` but the `i2c_dev_t` structure in [i2cdev.h](https://github.com/UncleRus/esp-idf-lib/blob/master/components/i2cdev/i2cdev.h) doesn't include a `clk_flags` member in its `cfg` structure, causing a compilation error.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

master

### Which build target have you used?

- [x] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp32c5
- [ ] esp32c6
- [ ] esp32c61
- [ ] esp32h2
- [ ] esp32p4
- [ ] esp8266
- [ ] other

### Component causing the issue

pcf8574

### Anything in the logs that might be useful for us?

```console
error: 'struct <anonymous>' has no member named 'clk_flags'
   79 |     dev->cfg.clk_flags = 0;
      |             ^
```

### Additional information or context

Add clk_flags to `i2cdev.h` structure to fix.
```c
struct
{
    gpio_num_t sda_io_num;
    gpio_num_t scl_io_num;
    uint8_t sda_pullup_en;
    uint8_t scl_pullup_en;
    uint32_t clk_flags;    // Add this field if needed
    struct
    {
        uint32_t clk_speed;
    } master;
} cfg;
```

### Confirmation

- [x] This report is not a question nor a request for drivers.

trombik, 2025-07-25:
Sorry, but would you wait until we finish the migration? See #719 and https://github.com/UncleRus/esp-idf-lib/discussions/720

trombik, 2025-07-25:
related repo: https://github.com/esp-idf-lib/i2cdev

trombik, 2025-07-28:
I made a PR but it's failing on ESP8266 for an unrelated reason.

You can try new [esp-idf-lib/pcf8574](https://github.com/esp-idf-lib/pcf8574). 

```sh
 idf.py add-dependency "esp-idf-lib/pcf8574^1.0.5" 
```

If your project depends on other components from `esp-idf-lib`, repeat the same step with different component names. The above command resolves dependencies and automatically uses the latest (and fixed) https://github.com/esp-idf-lib/i2cdev.

The code is the same. You do not need the whole `UncleRus/esp-idf-lib` any more.

https://github.com/esp-idf-lib/core has a list of all the components. [dps310](https://github.com/esp-idf-lib/dps310) has issues and I am fixing it.

trombik, 2025-07-30:
Closing. If you still find issues, open an issue at https://github.com/esp-idf-lib/pcf8574

closed 2025-07-30 by trombik